sstaub/Ticker #4: NOT working on arduino IDE 1.8.9

TasosR83, 2019-03-21:
It works on Arduino 1.8.7 but NOT on 1.8.9

sstaub, 2019-03-21:
Don't know what you mean, some error messages. It works for me on macOS.

TasosR83, 2019-03-21:
on Windows and Linux I have the message:

` Ticker timerMeasureBL(MeasureBL, 2 * 1000);
status
                              ^
exit status 1
'MeasureBL' was not declared in this scope`

The program is the following:
```
#include "Ticker.h"    
Ticker timerMeasureBL(MeasureBL, 2 * 1000);
```

sstaub, 2019-03-21:
Have you declared a callback function for MeasureBL?
You must also declare a prototype function for MeasureBL before setup.

sstaub, 2019-03-21:
You must declare the function or prototype before you declare the class member, so after #include "Ticher.h" and before Ticker timerMeasureBL(MeasureBL, 2 * 1000);

closed 2019-03-21 by sstaub