sstaub/Ticker #6: can i start a func which have a argument?

morestart, 2019-05-22:
RT

sstaub, 2019-05-22:
Did you try it?  I think it is not possible because of the function pointer type, but try it.

morestart, 2019-05-23:
> Did you try it? I think it is not possible because of the function pointer type, but try it.

Thanks for your replay, i will try it. If it is not possible, i thinks i can use global variable and  i will close this issue.

closed 2019-05-26 by morestart

pedzed, 2019-05-29:
The reason for my project not using this is for this primary reason. Otherwise, it looks like a nice library.

I also tried to use an anonymous function (lambda), but that didn't work as it does not seem to be supported by this library.

I opted for https://github.com/avandalen/VirtualDelay instead.

morestart, 2019-06-21:
> The reason for my project not using this library is for this primary reason. Otherwise, it looks like a nice library.
> 
> I also tried to use an anonymous function (lambda), but that didn't work as it does not seem to be supported by this library.
> 
> I opted for https://github.com/avandalen/VirtualDelay instead.

But it only support UNO, i what to find a ticker that can support esp8266 or esp32