sstaub/Ticker #9: Change the function pointer to a std::function for class member callbacks

markg85, 2019-07-14:
Hi,

Currently the callback logic seems to be focused on calling back global functions.
That is often what you want so it's definitely OK there!

However, i'm currently trying to incorporate ticker in a class of mine and want to call a member function in the callback. I can change the typedef you use to allow for that in my specific case, but it would be much nicer if it were just a std::function as this could then be done with std::bind.

For you, all you have to do is find/replace `fptr` with `std::function<void()>`.
And remove the typedef.

It still works how you're used to working with Ticker, but allows class member callbacks now. And i'm sure of that as that's how i'm using Ticker now :)

Cheers,
Mark

sstaub, 2019-07-14:
Sorry, your idea does't work with AVR microcontrollers because there is no support for c++ std libraries. Feel free to make your own fork for ARM microcontrollers.

markg85, 2019-07-14:
I don't think that's entirely right.
I'm using this on an ESP8266 (with the AVR toolchain!) and it works just fine with std::function.

sstaub, 2019-07-15:
The ESPs are also ARM.

pedvide, 2021-01-16:
Hey, I understand if you don't want to do this, but I think this should work (I can test it if you would be willing to merge it):
```
#ifdef __AVR__
typedef void (*fptr)();
#else
#include <functional>
using fptr = std::function<void()>;
#endif
```

sstaub, 2021-01-16:
Looks cool, but there are also the processor types like PIC which also doesn't support the std c++ lib. Maybe you differences between ARM, AVR and PIC. Please make a PR so we can test it before. Which c++ version supports <functional>, maybe this could also a problem.

sstaub, 2021-01-16:
the macro for identification of ARM processors is `__arm__` (lower case !)

pedvide, 2021-01-17:
There seems to be a standard way to check if a header is available:
```
#ifdef __has_include
#if __has_include(<functional>)
#include <functional>
using fptr = std::function<void()>;
#pragma message("functional detected!")
#else
typedef void (*fptr)();
#pragma message("functional not detected!")
#endif
#else
typedef void (*fptr)();
#pragma message("functional not detected!")
#endif
```

Unfortunately it seems it's not really working. Using Arduino or PlatformIO with a ESP8266 it says that functional is not detected, even though it exists.
I'm going to keep looking for ways to do this that don't involve hardcoding which architectures or boards have <functional>. Do you have a list of architectures you support? All the boards that can be used with the Arduino software?

sstaub, 2021-01-17:
The library can used with all microcontrollers supported by the Arduino framework. But only ARM 32bit microcontrollers like ESP,  STM32 supports the standard c++ library. I think there is no simple solution.

pedvide, 2021-01-17:
I've tested most boards that I can download using the Arduino board manager (including ESP8266, ESP32, and PIC with chipKIT), and this works:

```
#ifdef __has_include
  #if __has_include(<functional>)
    #pragma message("__has_include(<functional>)")
    #include <functional>
    using fptr = std::function<void()>;
  #else  // most boards
    #pragma message("NOT __has_include(<functional>)")
    #ifdef ESP8266 // ESP8266 has <functional>, but doesn't declare it
      #pragma message("ESP8266 detected!")
      #include <functional>
      using fptr = std::function<void()>;
    #else
      typedef void (*fptr)();
    #endif
  #endif
#else
  #pragma message("NOT __has_include")
  typedef void (*fptr)();
#endif
```

I've added indentation #pragma messages for debugging only.
Most boards define __has_include to be always false and have no support for `<functional>`, ESP8266 is the only one I could find that has support but doesn't say it. (ESP32 does say it).
I'm making a PR with this change.

sstaub, 2021-01-17:
Can you give also some short examples for using. I forgot <functional> is supported with c++11 maybe this is the problem with esp8266

pedvide, 2021-01-17:
Yes I'll add them to the PR

pedvide, 2021-01-17:
https://github.com/sstaub/Ticker/pull/27

closed 2021-01-25 by sstaub