sstaub/Ticker #10: Update repeat type to 32 bit

markg85, 2019-07-21:
Hi,

I'm using this library to fade in/out lights. So i'm having a very short timer (1ms) and a very large repeat value. Just for 1 second, that is already 1000. And with some experiments with a light that acts as a wakeup light (so slowly turn on over 30 minutes or so) means it has a value of: 1800000. Well above the 16 bit capacity.

Therefore i'm hoping the 16 bit restriction can be upped to 32 bit. I tried it locally, it works just fine!

Cheers,
Mark

sstaub, 2019-07-21:
Made a new release, please test it.

closed 2019-07-30 by sstaub

markg85, 2019-07-30:
I think it works, the code looks OK, but i'm now using a forked version of this great library (as per your request btw).

Another feature i added is calling a custom callback to indicate when the timer is finished, including the repeat count. You might be interested in that too. For me, that was valuable as i wanted to know when a timer was done to restart it. My usecase for that is a pulsating LED light where one timer run is from 0 to 100, the second timer run from 100 to 0. Then again and again...

sstaub, 2019-07-30:
In a previous version there was a function to check the state of the ticker, maybe I bring it back.