sstaub/Ticker #13: function isn't a type name

AlexFundorin, 2019-08-27:
When I'm declaring a timer inside main.cpp, everything is fine. Yet, as soon as I try to declare it inside a class, I get the following.
https://i.imgur.com/Kk6i9Uy.png
```
void blank(){};
Ticker timer1(blank, 1);
```

sstaub, 2019-08-26:
Were did you declare the function blank?

AlexFundorin, 2019-08-27:
I tried declaring it in several places:
main.cpp before loop
main.cpp before ticker include
inside the class (header file) right before the timer object
inside the same header file, but as a class function outside the class itself.

sstaub, 2019-08-27:
It sounds like a generell problem of nested classes. You should look in the web for it.

AlexFundorin, 2019-08-27:
Thanks for the suggestion.
For those, who'd encounter the same situation and doesn't need callback functionality, there's a [Chrono library](https://github.com/SofaPirate/Chrono), which is working fine in the same conditions.

closed 2019-08-27 by AlexFundorin