sstaub/Ticker #15: invalid use of non-static member function

WatsonMing, 2019-12-17:
Door *doorBuff = sensors[switchNo]->getDoor();

        DEBUG_SERIAL.println("switch");
        DEBUG_SERIAL.println(switchNo);

        if (doorBuff->getDoorStatus() == false)
        {
            doorBuff->openDoor();
            Ticker openDoorTicker(doorBuff->openDoorDelay, MOTO_TIME, 1, MICROS);
            openDoorTicker.start();
        }
        else
        {
            doorBuff->closeDoor();
            Ticker closeDoorTicker(doorBuff->closeDoorDelay, MOTO_TIME, 1, MICROS);
            closeDoorTicker.start();
        }

sstaub, 2019-12-17:
Don't know what you want to do with calling doorBuff->openDoorDelay you must call a function.

WatsonMing, 2019-12-20:
I want to  use the motor to open the door , so need function(openDoorDelay) stop the motor,*doorBuff is the Class(Door) Object pointer, function(openDoorDelay)  in the Class(Door)

WatsonMing, 2019-12-20:
I use polling instead of blocking as (function)Delay to perform the task of stopping the moto

sstaub, 2019-12-20:
Sorry, that is not possible. In the moment you can only use a function with none parameters.

closed 2019-12-29 by sstaub