sstaub/Ticker #17: Handling of time overflows

meltingSnowdrift, 2020-05-01:
Looking at the code, I do not see any way to continue correct behaviour when the timer overflows. In such a case, how will it behave? Considering the micros() value overflows approximately every 70 minutes and the millis() value overflows approximately every 50 days according to the Arduino documentation, does this mean that this library is unusable or will behave incorrectly for any use case in which the Arduino remains running for more than those durations? Because the default source of time information is now micros(), this would be a particularly serious problem because it is probably not infrequent for Arduino programs containing repeating events to run for longer than 70 minutes.

Although there may be no perfect solution to this, a majority of uses will probably use a ticker interval far shorter than either of those overflow intervals. Using this assumption, it should be possible to modify the library to function correctly across a time overflow.

Here is some code which I have used in a similar situation to solve this kind of problem. Sorry if it is wrong or not applicable here.
```
double timeSinceLastUpdate = updateTime - lastUpdateTime;
if(timeSinceLastUpdate < 0) //if a time overflow has occurred
{
    timeSinceLastUpdate = updateTime + (largestUnsignedLong - lastUpdateTime);
}
```

sstaub, 2020-05-01:
After an overflow, the library will also work correctly. The limitation are at MICROS the interval time
**maximum** is about 70 min, at MILLIS about 50 days. The counter will go to 0 when overflows and count up from there.

meltingSnowdrift, 2020-05-01:
If an interval starts before an overflow and ends after the overflow, will it work correctly? Will the delay between two events separated by such an interval be identical to that between two events separated by an interval that does not fall over an overflow?

If it does work, why? When micros() overflows, it goes back to 0. Wouldn't this instantly cause `micros() - lastTime` in your code to become very large, causing the timed event to occur at the moment of overflow regardless of the value of `timer`?

Sorry if I have perceived a problem where there is none and if I am misunderstanding your code.

closed 2020-06-03 by sstaub