sstaub/Ticker #20: How to use library with a function which needs to run as fast as possible

patfelst, 2020-06-02:
Hi there. I have a function which reads sensor data from a serial device, but I want this to run as fast as possible, ie not scheduled. Then I have other slower functions which I would like to schedule using your library.

The sensor data I read from serial takes approx 100ms to execute making my loop time about 10Hz. The other tasks need to run at 300ms intervals, and another one at 10 second intervals.

Is this possible and how would I structure the code? Thank you.

sstaub, 2020-06-03:
Make an ISR (Interupt) for the Serial device. The others can run with the Ticker library.

patfelst, 2020-06-03:
Thanks - I should have added, the serial device is actually via bluetooth classic SPP build into an ESP32. Not sure if I can use ISRs for that.

Let's say interrupts won't work, if I use Ticker to run my serial read function at 90ms interval, then each time around the loop .update() will be late, but would this effectively cause the function to run as fast as possible? Or should I just put my serial read code in the main loop() function, and use Ticker for the slower functions?

sstaub, 2020-06-03:
Put the serial code in the main loop(), the update of the ticker is very short. Only the execute of the callback need some time and depends how efficeint is your code. On an ESP32 you can also use the build in RTOS functionality.

patfelst, 2020-06-03:
Yes that sounds like a good approach. I've not used RTOS, would have to research that.

What I meant by .update() would be late: What happens when there is code in the main loop that takes longer to execute than the shortest Ticker callback interval? I'm assuming on the next .update() it runs immediately. 

e.g. I have a Ticker function 
`Ticker timer2(do_something, 50);` but there is code in the main loop that takes 100ms to execute.

sstaub, 2020-06-03:
Yes, on next update it runs immediately, ticker is measuring the time between the updates.
In this case your timer2 will execute of the after your code in main function is finished.
On RTOS you can handle it better if it is time critical.

patfelst, 2020-06-03:
thanks I will close now. I will try Ticker.

closed 2020-06-03 by patfelst

patfelst, 2020-06-05:
For your info, I used your library for my application last night. It was so easy, not one compile error. This was from Platform IO on Arduino platform for an ESP32. I looked at your library source code, so concise and clean and easy to understand.

I worked first time too - so thank you for this lightweight scheduler!