sstaub/Ticker #21: Does not work

Alf-R, 2020-06-06:
Hi, I was happy to find a Ticker libary but it does not work.
I'm using 
- Arduino 1.8.5
- NodeMCU 1.0 (ESP-12E)
- Windows
Ticker Lib is installed and found by Arduino
I loaded the Ticker example and got the compiler errors:

Arduino: 1.8.5 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\aremp\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\aremp\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\aremp\OneDrive\Dokumente\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:xtal=80,vt=flash,exception=legacy,ssl=all,eesz=4M2M,led=2,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -ide-version=10805 -build-path C:\Users\aremp\AppData\Local\Temp\arduino_build_297300 -warnings=none -build-cache C:\Users\aremp\AppData\Local\Temp\arduino_cache_493861 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mklittlefs.path=C:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-69bd9e6 -prefs=runtime.tools.mkspiffs.path=C:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.python3.path=C:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -verbose C:\Users\aremp\Documents\Arduino\libraries\Ticker\examples\Ticker\Ticker.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\aremp\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\aremp\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\aremp\OneDrive\Dokumente\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:xtal=80,vt=flash,exception=legacy,ssl=all,eesz=4M2M,led=2,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -ide-version=10805 -build-path C:\Users\aremp\AppData\Local\Temp\arduino_build_297300 -warnings=none -build-cache C:\Users\aremp\AppData\Local\Temp\arduino_cache_493861 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mklittlefs.path=C:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-69bd9e6 -prefs=runtime.tools.mkspiffs.path=C:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.python3.path=C:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -verbose C:\Users\aremp\Documents\Arduino\libraries\Ticker\examples\Ticker\Ticker.ino
Using board 'nodemcuv2' from platform in folder: C:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3
Using core 'esp8266' from platform in folder: C:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3
Detecting libraries used...
"C:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3/tools/sdk/include" "-IC:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3/tools/sdk/lwip2/include" "-IC:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\aremp\AppData\Local\Temp\arduino_build_297300/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11  -ffunction-sections -fdata-sections -fno-exceptions  -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0   -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU" -DLED_BUILTIN=2 -DFLASHMODE_DIO  -DESP8266 "-IC:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266" "-IC:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\variants\nodemcu" "C:\Users\aremp\AppData\Local\Temp\arduino_build_297300\sketch\Ticker.ino.cpp" -o "nul"
"C:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3/tools/sdk/include" "-IC:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3/tools/sdk/lwip2/include" "-IC:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\aremp\AppData\Local\Temp\arduino_build_297300/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11  -ffunction-sections -fdata-sections -fno-exceptions  -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0   -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU" -DLED_BUILTIN=2 -DFLASHMODE_DIO  -DESP8266 "-IC:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266" "-IC:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\variants\nodemcu" "-IC:\Program Files (x86)\Arduino\libraries\Ticker" "C:\Users\aremp\AppData\Local\Temp\arduino_build_297300\sketch\Ticker.ino.cpp" -o "nul"
"C:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3/tools/sdk/include" "-IC:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3/tools/sdk/lwip2/include" "-IC:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\aremp\AppData\Local\Temp\arduino_build_297300/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11  -ffunction-sections -fdata-sections -fno-exceptions  -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0   -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU" -DLED_BUILTIN=2 -DFLASHMODE_DIO  -DESP8266 "-IC:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266" "-IC:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\variants\nodemcu" "-IC:\Program Files (x86)\Arduino\libraries\Ticker" "C:\Program Files (x86)\Arduino\libraries\Ticker\Ticker.cpp" -o "nul"
Generating function prototypes...
"C:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3/tools/sdk/include" "-IC:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3/tools/sdk/lwip2/include" "-IC:\Users\aremp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\aremp\AppData\Local\Temp\arduino_build_297300/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11  -ffunction-sections -fdata-sections -fno-exceptions  -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0   -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU" -DLED_BUILTIN=2 -DFLASHMODE_DIO  -DESP8266 "-IC:\Users\aremp\AppData\Local\Arduino15\packages\esp826 [... cut ...]
Ticker:43: error: 'class Ticker' has no member named 'counter'

   Serial.println(timer2.counter());

                         ^

C:\Users\aremp\Documents\Arduino\libraries\Ticker\examples\Ticker\Ticker.ino: In function 'void printCountdown()':

Ticker:48: error: 'class Ticker' has no member named 'counter'

   Serial.println(5 - timer3.counter());

                             ^

Bibliothek Ticker in Version 1.0 im Ordner: C:\Program Files (x86)\Arduino\libraries\Ticker  wird verwendet
exit status 1
'MILLIS' was not declared in this scope

sstaub, 2020-06-06:
Maybe there is another Ticker library installed for ESP32. The compiler does not use this Ticker library.

Alf-R, 2020-06-06:
you're right, thank you

closed 2020-06-06 by Alf-R