sstaub/Ticker #25: Member function as callback parameter

vindolin, 2020-09-26:
I'm a C++ noob and I'm having a hard time figuring out how to supply a member function as the callback parameter for the Ticker.

This code:
```c++
void Meep::myCallback() {
}

void Meep::init() {
    Ticker tickerMeep = Ticker(myCallback, 5000);
}

```
Gives me the error message ```error: no matching function for call to 'Ticker::Ticker(<unresolved overloaded function type>, int)'```

Do you have any ideas how this could be achieved?

Thank you for the nice library which I'm using in almost every project now.

sstaub, 2020-09-26:
In the moment this not possible. I think about a solution but the problem is the AVR compatibilty because there is no std library.

sstaub, 2020-09-26:
I have no idea, nested classes are very special in handling.

closed 2020-10-12 by sstaub