sstaub/Ticker #26: Unstable when doing Serial.print(x) of an integer.

gibo77, 2021-01-02:
Unstable when doing Serial.print(x) of an integer. Unstable means the printf output is offset. The wifi sometimes will not connect. Wifi functions are not working.
If I remove the '\`+ HighLevelStatus2secCount`, everything goes back to working program.
`int HighLevelStatus2secCount=0;
int LowLevelStatus2secCount=0;
void RunPeriodicallyTicker();
Ticker timer5(RunPeriodicallyTicker, 2500, 0, MILLIS); // RunPeriodicallyTicker function will run every 2.5 seconds.
void setup() { 
//Wifi functions
  timer5.start();
}

void loop() {
//Body of program
}
void RunPeriodicallyTicker() {
  if (LevelStatus == HIGH) { //Float is High and 120VAC is not applied
      HighLevelStatus2secCount++;
      LowLevelStatus2secCount = 0;
      Serial.println("Float Switch Up! " + HighLevelStatus2secCount);
      if (HighLevelStatus2secCount>40){ //900 * 2.5seconds is equal to 37minutes roughly
        HighLevelStatus2secCount=0; //so that the number will not get too big and crash the microprocessor
        ProbableSumpPumpFailure(); //We need to send data to AWS Gateway that there could be Sump Pump Failure due to level did not go down within estimated time.
        
      }
    }`

sstaub, 2021-01-02:
missing timer5.update()
what does timer5?

gibo77, 2021-01-02:
There is timer5.update() in the loop(). It runs but unstable and I forgot to include that it crashes to stack trace below after few minutes.
```
Exception (0):
epc1=0x40234475 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffed60 end: 3fffffb0 offset: 01a0
3fffef00:  00000000 3ffee0b0 3ffee088 402102d0  
3fffef10:  40104e35 3ffeaf96 4023dd8f 3ffec808  
3fffef20:  3ffeaf80 3fffdcc0 3ffe8de0 3ffe8de0  
3fffef30:  00000040 3ffec808 00000000 3ffe84f8  
3fffef40:  4023d64f 3fffdab0 00000000 40205b1b  
3fffef50:  3ffe8de0 40000f49 3fffdab0 40000f49  
3fffef60:  40000e19 40001878 00000004 00000000  
3fffef70:  3fffff10 aa55aa55 0000008f 401047d5  
3fffef80:  401047db 00000004 00000000 9ee14020  
3fffef90:  4010000d 2a5142e5 7d912cbe ffc0c141  
3fffefa0:  00000000 3fffef3c 00000000 3ffffef8  
3fffefb0:  3fffffc0 00000000 00000000 feefeffe  
3fffefc0:  feefeffe feefeffe feefeffe feefeffe  
3fffefd0:  feefeffe feefeffe feefeffe feefeffe  
3fffefe0:  feefeffe feefeffe feefeffe feefeffe  
3fffeff0:  feefeffe feefeffe feefeffe feefeffe  
3ffff000:  feefeffe feefeffe feefeffe feefeffe  
3ffff010:  feefeffe feefeffe feefeffe feefeffe  
3ffff020:  feefeffe feefeffe feefeffe feefeffe  
3ffff030:  feefeffe feefeffe feefeffe feefeffe  
3ffff040:  feefeffe feefeffe feefeffe feefeffe  
3ffff050:  feefeffe feefeffe feefeffe feefeffe  
3ffff060:  feefeffe feefeffe feefeffe feefeffe  
3ffff070:  feefeffe feefeffe feefeffe feefeffe  
3ffff080:  feefeffe feefeffe feefeffe feefeffe  
3ffff090:  feefeffe feefeffe feefeffe feefeffe  
3ffff0a0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0b0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0c0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0d0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0e0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0f0:  feefeffe feefeffe feefeffe feefeffe  
3ffff100:  feefeffe feefeffe feefeffe feefeffe  
3ffff110:  feefeffe feefeffe feefeffe feefeffe  
3ffff120:  feefeffe feefeffe feefeffe feefeffe  
3ffff130:  feefeffe feefeffe feefeffe feefeffe  
3ffff140:  feefeffe feefeffe feefeffe feefeffe  
3ffff150:  feefeffe feefeffe feefeffe feefeffe  
3ffff160:  feefeffe feefeffe feefeffe feefeffe  
3ffff170:  feefeffe feefeffe feefeffe feefeffe  
3ffff180:  feefeffe feefeffe feefeffe feefeffe  
3ffff190:  feefeffe feefeffe feefeffe feefeffe  
3ffff1a0:  feefeffe feefeffe feefeffe feefeffe  
3ffff1b0:  feefeffe feefeffe feefeffe feefeffe  
3ffff1c0:  feefeffe feefeffe feefeffe feefeffe  
3ffff1d0:  feefeffe feefeffe feefeffe feefeffe  
3ffff1e0:  feefeffe feefeffe feefeffe feefeffe  
3ffff1f0:  feefeffe feefeffe feefeffe feefeffe  
3ffff200:  feefeffe feefeffe feefeffe feefeffe  
3ffff210:  feefeffe feefeffe feefeffe feefeffe  
3ffff220:  feefeffe feefeffe feefeffe feefeffe  
3ffff230:  feefeffe feefeffe feefeffe feefeffe  
3ffff240:  feefeffe feefeffe feefeffe feefeffe  
3ffff250:  feefeffe feefeffe feefeffe feefeffe  
3ffff260:  feefeffe feefeffe feefeffe feefeffe  
3ffff270:  feefeffe feefeffe feefeffe feefeffe  
3ffff280:  feefeffe feefeffe feefeffe feefeffe  
3ffff290:  feefeffe feefeffe feefeffe feefeffe  
3ffff2a0:  feefeffe feefeffe feefeffe feefeffe  
3ffff2b0:  feefeffe feefeffe feefeffe feefeffe  
3ffff2c0:  feefeffe feefeffe feefeffe feefeffe  
[... 68 lines cut ...]
3ffff710:  c7ee43df c956d0f4 8fd1b8ed 775e2ee2  
3ffff720:  8182d2cd 9b1b6819 5286e92e 49d3ab47  
3ffff730:  f2fd139d 30aaf226 a2e02747 00000080  
3ffff740:  03030000 010000f2 1fee2160 000021c7  
3ffff750:  90e3174e 02f6977d ebb3ef4a 45839d56  
3ffff760:  bddf0000 97b0275d a8c02bc0 5acca9cc  
3ffff770:  30c0acc0 2fc02cc0 afc023c0 adc0aec0  
3ffff780:  28c009c0 27c024c0 14c02dc0 13c00ac0  
3ffff790:  32c025c0 31c02ec0 2ac004c0 29c026c0  
3ffff7a0:  0f009c00 0ec005c0 9dc0a0c0 9dc09cc0  
3ffff7b0:  3d002f00 a1003c00 12c003c0 35c008c0  
3ffff7c0:  ce0fcfac 84261105 38c21214 2febe12a  
3ffff7d0:  efef3b3d 6f9b8938 45b54c3b dc468a06  
3ffff7e0:  4b1a6f56 bcff4f4b 23edeead 1c1f6d75  
3ffff7f0:  f5a77c6f a5799d0b f8379f15 62cdc772  
3ffff800:  5d2b72a2 e85545ee 1099355c 40158be0  
3ffff810:  c627a7e0 12003341 73638fde 153a3a27  
3ffff820:  72823589 1edb1b7d f3c453c3 6d1c5384  
3ffff830:  81431ac7 8e9edc49 60de7f14 f0e3322a  
3ffff840:  38663ec3 dd55d28b 06e73407 17a1fffb  
3ffff850:  e0e84ecd 87210ab6 1564f712 60ade6cc  
3ffff860:  4b088fc8 80149b57 b3aa0b0a 01215bcf  
3ffff870:  d79c47c8 4e7c1dfc d5f7a8a7 5fa5126d  
3ffff880:  5893ae48 76e257b4 cd8ò†„[0E]F
```

sstaub, 2021-01-02:
Sorry I don't understand the problem, it looks that this is not a problem of the ticker library.
It looks like a problem with the HighLevelStatus2secCount variable.

Quercuz, 2021-01-10:
Try: Serial.println("Float Switch Up! " + **_String_**(HighLevelStatus2secCount));

closed 2021-01-16 by sstaub