sstaub/Ticker #29: Functional callbacks work on Tensilica (ESP8266/32) too, not only ARM.

vindolin, 2021-01-26:
I tried your FunctionalARM.ino example and it also works fine on the Espressif Tensilica Chips.

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

B::func: true.
A::func: true.
A::func: false.
B::func: true.
C(): 4.
func.
B::func: true.
C(): 4.
func.
A::func: true.
A::func: false.
C(): 4.
func.
A::func: true.
A::func: false.
B::func: true.
A::func: true.
A::func: false.
B::func: true.
C(): 4.
func.
B::func: true.
```

sstaub, 2021-01-26:
For comments please use the discussion forum, but it is good to that the examples work, thanks to @pedvide

closed 2021-01-26 by sstaub

vindolin, 2021-01-26:
Then the readme misled me, as it reads "added support for functional callbacks, **only** for ARM devices, e.g.".

sstaub, 2021-01-26:
I will change the Readme