sstaub/Ticker #32: Endless repeat

mikmade, 2021-03-05:
Hello,
First of all, Thanks for this great library. 
I would like to use it with an endless repeat, but after reading your code I think I found a minor bug (maybe I am wrong).
The problem I am seeing is that after an overflow of the counter variable, the loop will stop because repeat - counter = 1 will become true. 
Of course this bug will only occur after a quite long time (depending on the ticker frequency), but is still there (and relatively easy to fix). 
As I said maybe I made a mistake and you are taking care of somewhere I couldn't find.

sstaub, 2021-03-05:
Yes, maybe that can occur. If you have fix, let me know or make a PR.

sstaub, 2021-03-05:
No, the Ticker doesn't stop but the counter will reset to 0 and then growing up again.

mikmade, 2021-03-05:
The counter will never come to 0, because it will stop at 0xFFFFFFFF. ( 0 - 0xFFFFFFFF == 1)

sstaub, 2021-03-05:
It works as expected, I tested it simplifying changing `uint32_t counts;` to `uint8_t counts;`
and following code:
```cpp
#include "Arduino.h"
#include "Ticker.h"

void printMessage();

bool ledState;

Ticker timer1(printMessage, 500);

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(9600);
  timer1.start();
  }

void loop() {
  timer1.update();
  }

void printMessage() {
  if (ledState) {
    digitalWrite(LED_BUILTIN, HIGH);
    ledState = false;
    }
  else {
    digitalWrite(LED_BUILTIN, LOW);
    ledState = true;
    }
  Serial.print("Time ms ");
  Serial.print(millis());
  Serial.print(" Count ");
  Serial.println(timer1.counter());
  }
```
results in serial monitor:
[image]

mikmade, 2021-03-05:
Testing it by changing the variable type creates a new situation, because now there is an implicit cast to uint32_t and 0x00 - 0xFF => 0xFFFFFF01 == 1 => false
If you test it by changing the initial value of counts to something like 0xFFFFFFF0 then you can see the problem.

sstaub, 2021-03-05:
There is no implicit cast, only when you use counter(), but this doesn't affect the variable counts. The initial value is always 0.

mikmade, 2021-03-05:
there is one in the comparison. If you want to test it change counts = 0;  in the start method to counts = 0xFFFFFFF0 and test it with your example (counts has to be a uint32_t of course) 
I also send you a pull request with the change of having an if statement around the counts++;
I don't want to annoy you any longer. If you don't belief me it's fine. I will use it with the modification and wanted to share it.
Thanks for the code.

sstaub, 2021-03-05:
Try to change repeats and counts to uint8_t you will see there is no problem with casts.
Is this a theoretical thing or is there a real practical problem?

mikmade, 2021-03-05:
It is a real practical problem, but only after a really long time, so probably not relevant for most projects, emphasis on most.
Try it with your own example and your original library with one line changed (so you don't have to wait a year)
```c
void Ticker::start() {
	if (callback == NULL) return;
	if (resolution == MILLIS) lastTime = millis();
	else lastTime = micros();
	enabled = true;
	counts = 0xFFFFFFF0;  // This line is changed
	status = RUNNING;
}

sstaub, 2021-03-05:
Ok, I see it.

sstaub, 2021-03-05:
Solved with 4.1 release

closed 2021-03-05 by sstaub