sstaub/Ticker #37: library name

stef-ladefense, 2021-03-24:
hello, can you name your library STicker for example?

Ticket already exists internally with esp8266 and esp32 and it makes incompatibilities to use one or the other.
and it is counterproductive which is the same name, it breaks the updates of the kernel and your library too.

if you think (like me) that your library is better and can replace their internal libraries request a replacement.

thank you very much for your excellent work

sstaub, 2021-03-24:
No way, this will cause problems for all other users. I know about the problem.

closed 2021-03-24 by sstaub

stef-ladefense, 2021-03-24:
then explain how not to have a confluence with the internal ESP libraries.
when I declare #include "Ticker.h" it's not yours to use.

Several libraries found for "Ticker.h"
Used: C: \ Users \ stephane \ AppData \ Local \ Arduino15 \ packages \ esp8266 \ hardware \ esp8266 \ 2.7.4 \ libraries \ Ticker
Not used: C: \ Users \ stephane \ Documents \ Arduino \ libraries \ Ticker-master

how to do ?