sstaub/Ticker #38: conflict

stef-ladefense, 2021-03-24:
then explain how not to have a confluence with the internal ESP libraries.
when I declare #include "Ticker.h" it's not yours to use.

Several libraries found for "Ticker.h"
Used: C: \ Users \ stephane \ AppData \ Local \ Arduino15 \ packages \ esp8266 \ hardware \ esp8266 \ 2.7.4 \ libraries \ Ticker
Not used: C: \ Users \ stephane \ Documents \ Arduino \ libraries \ Ticker-master

how to do ?

sstaub, 2021-03-24:
I don't know if the internal library of espressif is used, if not you can delete it or you rename the ticker library for your purpose.

stef-ladefense, 2021-03-24:
as I already wrote
1) deleting the internal ESP8266 or ESP32 library is not a good solution, it breaks the reference and it will come back to the next update version.
2) renaming your library is not enough, because you must also rename all the internal declarations, but it is not a good solution either because it will be broken during your next update version.

this is why I proposed for it to be compatible without hack to rename it to STicker

sstaub, 2021-03-25:
The library has thousands of users, so definitly no.

closed 2021-03-25 by sstaub

stef-ladefense, 2021-03-25:
it's a shame to cut yourself off from the ESP community