sstaub/Ticker #39: error in readme.md

stef-ladefense, 2021-03-24:
examples 

Tickers timer(blink, 1000); // calls function blink() every second, internal resolution is **micros**, running endless
Tickers timer(blink, 1000, 5); // calls function blink() every second, internal resolution is **micros**, only 5 repeats
Tickers timer(blink, 1000, 0, MILLIS); // calls function blink() every second, internal resolution is millis, running endless
Tickers timer(blink, 1000, 0, MICROS_MICROS); // calls function blink() every 1000 microsecond, internal resolution is micros, running endless

sstaub, 2021-03-24:
internal resolution is micros() as standard

stef-ladefense, 2021-03-24:
you write 
mode set the interval resolution to **MILLIS (default)** or MICROS
it's wrong then ?

sstaub, 2021-03-25:
It was MICROS and I correct the readme.

closed 2021-03-25 by sstaub

stef-ladefense, 2021-03-25:
Good ! merci