sstaub/Ticker #40: MICROS and MICROS_MICROS

stef-ladefense, 2021-03-25:
I can't find an explanation between the two in your explanations.

if I understood the code correctly, by default "MICROS"
write "Ticker timer (printCounter, 1000, 0);"
& "Ticker timer (printCounter, 1000, 0, MICROS);"
it's identical, and 1000 is x1000 internally, so 1 second

Ticker timer (printCounter, 1000, 0, MICROS_MICROS);
it is 1000 microsecond (1ms) in this case, correct?
(with 4µs resolution for ATMega328 @ 16Mhs)

therefore MICROS, the duration is multiplied by 1000 internally,
if we are not careful, and logically we say to ourselves, since the resolution is microsecond, we make the mistake of thinking that 1000 is 1 millisecond

the most logical would have been to be in MILLIS by default (exactly like delay), and not to ask any questions.
and specify only MICROS for microsecond times but not multiply by 1000 internally.

your chosen solution does not make sense to me, but I am not criticizing.
it should just be explained correctly in the documentation

thank you again for your work

sstaub, 2021-03-25:
If you want to discuss things, because you don't read the docs, use the discussion forum.
The explanation you can find under 'New in v2.0' says micros() is used when the interval time is under 70min.

closed 2021-03-25 by sstaub

stef-ladefense, 2021-03-25:
when I look for the syntax of a function, I expect to have the options in the same place, and not to read the set of revisions which cancel each other to clear up the use.

so yes I'm boring, but at least I accept constructive criticism

AlesSt, 2021-07-05:
Sorry to steal this closed issue but it relates :)

Great library, super helpfull :) 

But 
 - i dont understand the difference between MICROS and MICROS_MICROS in example says 1000us is 1 s .... 1000us is 1ms => but i guess only default internal unit is us timer is still in ms? => if so then in examples it should be written 1000 ms is 1s :) 

So if i get this correctly MICROS => internally uses microseconds but timer is in miliseconds and MICROS_MICROS => internally same microseconds and also timer uses microseconds?

BR
Ales