sstaub/Ticker #44: Work with ticker without calling ticker.update()

vginside, 2021-05-24:
Hi guys I need your help, I needed to use ticker so initially i was using "https://github.com/esp8266/Arduino/tree/master/libraries/Ticker" but then i need a ticker for microsec which this library did not supported, so i came across by library another library by sstaub "https://github.com/sstaub/Ticker". Now this library needs a must call (object).update() called in loop other wise callback function is not called, unlike earlier there was no need for function like update in main loop... For example i need a ticker to blink LED while wifi is not connected and it is in loop WL_CONNECTED in void setup() and code has still not entered void loop(). So ticker callback is not called since it (Object).update() is in main loop. Is not possible to not use update like original ticker.h (ofc i dont only need to blink LED so dont give me an alternative to help me blink LED.)
This ticker has lot of advantages over old ticker and i need to use this library. So how can i get a work around? Thanks in Advance!
 Best Regards!

sstaub, 2021-05-24:
I think the ESP library use hardware timers so no manually update is needed.

vginside, 2021-05-24:
Your library is really good so many added benefits but only disadvantage is calling update(). So we cant really use it like timer Interrupt and get precise timings out of it! Is it not possible to make your library work with something like ESP library? It would be really helpful! Awaiting your reply.

Best Regards!

sstaub, 2021-05-24:
Sorry, but calling update() is necessary and can not changed. My library only uses the millis() / micros() timer instead of hardware timers which gives conflicts with  e.g. PWM ... when using small microcontrollers like the Arduino UNO.

vginside, 2021-05-24:
> Sorry, but calling update() is necessary and can not changed. My library only uses the millis() / micros() timer instead of hardware timers which gives conflicts with e.g. PWM ... when using small microcontrollers like the Arduino UNO.

Hi again! I appreciate the hard work.. If that that is the case maybe creating specifically a new library specifically for ESP MCU's named like 'ESPTicker.h'?

sstaub, 2021-05-24:
Did you mean a renamed library, or specific for esp with hardware timers.

vginside, 2021-05-24:
I mean if you can creating a one more new exclusive library for ESP devices only named like "ESPTicker.h" or something ,  which will hardware timers and have all the functions from your library so it dont need update() function. 
  You can also code it with such way if devices used are ESP it will use hardware timer like ESP library but if non ESP devices you need to write update() function.(frankly this will be very confusing rather i prefer exclusive library for ESP devices).
Sorry if i'm asking way too much :)

sstaub, 2021-05-24:
Sorry but I can't do it because I use no ESP.

closed 2021-05-24 by sstaub