sstaub/Ticker #45: Ticker compile error with ESP8266 in Arduino

drc38, 2021-06-07:
Hi, when I compile using an Arduino board such as an Uno the sketch with your Ticker library compiles without error. However, when I switch to an ESP8266 board such as a Wemos mini I get the following error `Ticker.cpp:27:15: error: expected constructor, destructor, or type conversion before '(' token
   27 | Ticker::Ticker(fptr callback, uint32_t timer, uint32_t repeat, resolution_t resolution) {` . Is there something I need to specify in the sketch to work with ESP boards?
Thanks

sstaub, 2021-06-07:
There is also library called Ticker in the esp framework.

drc38, 2021-06-07:
Initially that was the problem, however I renamed your library and relevant files to newTicker to avoid the conflict. It then throws the error above, prior to this the compiler was trying to use the esp framework Ticker.

drc38, 2021-06-07:
Sorry, just realised I had not updated in the `#include` in the .cpp to newTicker. Now it is compiling without error. Is there a way to reference your Ticker without renaming it?

sstaub, 2021-06-07:
No.

closed 2021-06-07 by sstaub