sstaub/Ticker #55: How to identify if the Ticker has already been triggered at least once?

georgevbsantiago, 2022-09-03:
My goal is to identify whether Ticker has already been called to handle its range.
That is, identifying the first ticker trigger and being able to manipulate the interval with interval().

Example: I would like a callback to fire at the very beginning of the loop execution and then every 300 seconds (5 minutes).
I currently use a global variable for this, but I would like to know if Ticker has any features or can implement some solution so that the global variable is not needed.
Ex:

```
Ticker uplink_TICKER(uplink_function, 10, 0, MILLIS);

bool first _uplink = true;

main { ... }

loop {

uplink_TICKER.update();

            if(first _uplink == true) {
            
                        first _uplink = false;
            
                        uplink_TICKER.resume();

                        uplink_TICKER.interval(300000);          
            }

}
```

sstaub, 2022-09-03:
There is no internal solution. Maybe you create two tickers with the first to shoot only once.

sstaub, 2022-09-03:
Forgot, there is also counter() function, which gives you back the ticks count.

georgevbsantiago, 2022-09-05:
Would the approach below be viable? If the user set `true`, the 1st callback would only be at 300000 millis. But if the user set `false`, it would make the 1st callback execute immediately (at 0 millis) and the rest of the callbacks every 300000 millis.

`Ticker uplink_TICKER(uplink_function, 300000, 0, MILLIS, true);`
[image]

`Ticker uplink_TICKER(uplink_function, 300000, 0, MILLIS, false);`
[image]

At first, I don't see a problem because there are situations where the user wants the callback to be triggered, but the resource is not used by another callback.
Example: The user wants the display to show an "ERROR in Execution" message with TICKER_1 (which triggers a callback immediately), but wants this message to stay for 8000 millis, using the logic of RUNNING or STOPPED to trigger a TICKER_2. if(TICKER_1.state() == STOPPED) { TICKER_2.resume(); }

`Ticker uplink_TICKER_1(uplink_function, 8000, 0, MILLIS, false);`
[image]

sstaub, 2022-09-05:
Why didn't you trigger the callback function manually?

georgevbsantiago, 2022-09-16:
My intention with the proposal would be to solve two problems:
1) Configure a Ticker `(function, 8000, 0, MILLIS, false)` and immediately trigger the function registered in the Ticker callback at the very beginning [counter == 0] in the first iteration of void loop() and the following a every 8000 millis.
2) No need to handle interval() and counter() to allow a semaphore approach to using the Display (ie only one Ticker at a time, showing messages on the Display).

The first proposal I described above. The second is described below:

I needed to apply an approach that respected the idea of priority and semaphore to use the 16x02 Display with only one Ticker at a time.
If I run a function without TICKER, I can't structure priority and semaphore logic. Ticker helps in this objective by manipulating the states (STOPPED, RUNNING, PAUSED).

Example:
[image]

However, in Ticker's current approach, I can only handle the RUNNING state with 2 tickers. And my idea would be to manipulate it with just 1 ticker, as in the example below:
[image]

Sorry if I'm saying something stupid. I'm just a Maker who uses and really likes your library. Thank you for developing this fantastic library.

georgevbsantiago, 2022-11-10:
Maybe the `queueFunction` example from the H4 library will help something about this issue:

https://github.com/philbowles/H4#introduction

> With one exception (queueFunction) all tasks start after the first specified time interval and not immediately. Using the infinite task "every(1000..." will invoke the first instance of user callback at Tstart + 1sec (1000 mS).

Thanks for developing Ticker

closed 2025-07-03 by sstaub