sstaub/Ticker #57: Versions 4.x.x error compiling for Arduino Due

5chufti, 2023-04-17:
Hi,
after updating from 3.2.0 to current version my sketch failed compiling.
I am using Arduino IDE 1.8.19 and Arduino SAM Boards package 1.6.12.
The errormessages are complaining about macro swap being called with too less arguments.

But even the examples coming with the library are not compiling either.

sstaub, 2023-04-17:
I don't know what the ticker library have to do with a swap macro. Maybe a problem with the std:function

5chufti, 2023-04-17:
if I change 
#if defined(__arm__) || defined(ESP8266) || defined(ESP32)
to
#if defined(ESP8266) || defined(ESP32)

at least my sketch and the Ticker example compile, the FunctionalARM still fails at least because Due does not support Serial.printf()

sstaub, 2023-04-17:
Maybe Due use an old compiler version. Sorry but you must use your modified version.