sstaub/Ticker #60: Build failure with STM32 Blue Pill board

ukoda, 2024-02-20:
When I try to build with version 4.4.0 for a STM32F103C8 Blue Pill board I get the error:
```
Compiling .pio/build/bluepill_f103c8/lib6e0/Ticker/Ticker.cpp.o
In file included from src/main.cpp:19:0:
.pio/libdeps/bluepill_f103c8/Ticker/Ticker.h:56:7: error: expected nested-name-specifier before 'fptr'
 using fptr = std::function<void()>;
```
Followed by the normal errors you would expect after that.

The problem may be at line 54 of Ticker.h, `#if defined(__arm__) || defined(ESP8266) || defined(ESP32)`.  I changed that area to just  `typedef void (*fptr)();` and then everything builds and runs ok.

I have not dug any deeper as that change is enough to keep me going but I figured you may wish to review the issue so it does not catch other people out.

If you need more detail just let me know what.

sstaub, 2024-02-20:
Maybe this is a problem with the compiler version for the Blue Pill.

ukoda, 2024-02-20:
@sstaub That is quite likely as I also see the warning:
```
In file included from src/main.cpp:19:0:
.pio/libdeps/bluepill_f103c8/Ticker/Ticker.h:153:28: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
  resolution_t resolution = MICROS;
```
I'm building under PlatformIO with the Arduino frame work.  The terminal output doesn't show what compiler version it is using and I can't see where to find that info in the project settings.

closed 2025-07-03 by sstaub